LibreRouterOrg/board #13: ANT3 connector - no signal

nicopace, 2017-11-09:
The ANT3 connector on the primary QCA9558 radio is not producing any signal on TG sample board.
ANT1 produces 16dBm signal, and ANT2 produces 6dBm signal on TG sample board.
Elektra reports signal present on all ANT connectors, so the ANT3 problem looks like an individual board fault.
Note that to get max power output, the wifi Country parameter has to be set to non-default value eg AU
TxPower max on QCA9558 displays in "iw dev" output as "31dBm" with Country set to AU.

tgillett, 2017-11-09:
Elektra's measurements of TxPower with default wifi config settings in the test firmware are:
  4 mW at ANT 1
  1 mW at ANT 2
  0.05 mW at ANT 3

On Terry's test board the measurements for default settings (Country CN, TxPower 23, iw dev reports power level of "20") are:
  ANT-1    4.5dBm    2.5mW
  ANT-2    -10dBm    0.1mW
  ANT-3    No signal at all

With the config changed to Country AU and TxPower set to default, iw dev reports power level of "30", the measured power levels are:
  ANT-1   14.5dBm   30mW
  ANT-2   0.5dBm      1.2mW
  ANT-3   No signal at all.

ANT-1 level is probably OK given that the board has not had individual RF calibration.
This is the level that is expected directly from the QCA9558 SoC.
Need to check that the PA chip is actually working to raise the level to the spec 23dBm.

ANT-2 level is reduced probably due to the output filter component error as shown on schematic/

ANT-3 signal is not working on either test board which probably indicates some component or circuit issue that is not yet obvious.

spiccinini, 2018-11-21:
Fixed in CoreBoard v0.2 (That has only ANT-1 and ANT-2)

closed 2018-11-29 by spiccinini