LibreRouterOrg/board #14: PoE is not working

nicopace, 2017-11-09:
PoE is not working

tgillett, 2017-12-26:
Can't be effectively tested in the initial prototype board due to errors aroung RJ45 connector wiring.
Moved to Phase 2 testing to check in board re-design.

dragino, 2018-01-24:
Found circuit issue and will be fixed in the 9558 Mega Board v1.0

spiccinini, 2018-11-21:
PoE is working, tested with Mega Board v1.1 and Dragino external PoE injector from 12 to 32V, and 2m CAT5 cable.

Some notes
* PoE uses only ETH pins 5 (VCC) and 8 (GND). If we want to use long ETH cables we should use more pins.
* It should be compatible with ubiquity PoE (VCC in 1,2,4,5, GND in 3,6,7,8).
* Power consumption with one mPCI full radio Tx goes from 10.5W@12V to 11.5W@32V with PoE. Powering the board directly with a power supply consumption with one mPCI full ratio tx: 8.8W@12V to 11.8W@32V.
* T1 magnetics (QT48A03) gets very very hot in one minute when doing **PoE passthrough** to another LR. QT48A03 datasheet doesn't say it is PoE compliant.

spiccinini, 2018-12-03:
I did more power consumption measurements:

* From factory with R1, R2, R4, R5:
    * 3.2W@12V IDLE wifi down without ethernet
    * 4.4W@12V **POE** IDLE wifi down without ethernet -> heat in magnetics, not very hot 
    * 8.6W@12V 1 mPCI full tx

* Without R1, R2, R4, R5:
    * 3.3W@12V **POE** IDLE wifi down without ethernet -> some heat in magnetics, not cold but not hot 
    * 10.5W@12V **POE** 1 mPCI full tx  ->  overheats magnetics

So, removing R1,R2,R4 and R5 improves but I think it's not good enough for production or at least more tests should be made. I don't have a gigabit PoE injector that uses more pairs (Dragino's PoE uses only two wires) I will buy one and test with a rework with the  circuit attached (changes in blue, using the four pairs, removing R1, R2, R4 and R5).
[magnetics.pdf](https://github.com/libremesh/librerouter/files/2640512/magnetics.pdf)

spiccinini, 2018-12-12:
Measurements https://github.com/libremesh/librerouter/wiki/PoEPT-(Pass-through)-heat-vs-voltage

spiccinini, 2018-12-18:
PoE is working so I am closing this issue. The heat issues are in #23

closed 2018-12-18 by spiccinini