LibreRouterOrg/board #16: LEDs for 5GHz WLAN Operation

nicopace, 2017-11-09:
LEDs for 5GHz WLAN are both lit with one board installed.

The 5G WLAN LEDs are shown on schematic pp 11 as being driven by signals:
5G_WLAN_LED_0
5G_WLAN_LED_1
These signals appear on schematic pp 15 and 16 respectively as connected to the PCIe connectors Pin 44.
The circuit looks correct. There is no obvious reason why both LEDs should light when only one PCIe board is installed.
This may be a fault on the sample board.
TODO - check operation on other sample boards.

elektra42, 2017-11-14:
The reason for this behavior is that the pads LED+ of both LEDs are bridged on the pcb. (You can measure 0 Ohm resistance between the pads.)

tgillett, 2017-12-26:
Edwin has investigated and discovered that there is an error in the schematic that ties both LED lines together,
Moved to Phase 2 testing to check that it is corrected in the board re-design.

spiccinini, 2019-05-15:
Fixed at least in MegaBoard >=v1.1

closed 2019-05-15 by spiccinini