LibreRouterOrg/board #2: Rx Sensitivity

nicopace, 2017-11-09:
Rx Sensitivity of the QCA 9558 SoC seems to be low, and varies between ANT connectors on some sample boards.

Testing on TG sample board indicates that Rx Sensitivity is ~10dB lower than that of a comparison WR703 device, but is the same (within 2dB) for all ANT connectors.
Testing on Elektra sample boards indicates ~20dB below comparison level, and with ~6dB variability between ANT connectors.

This is not an RF calibration problem. It is likely to be a problem with components in the RF output circuits around the PA, LNA and T/R switch components.

dragino, 2018-01-24:
In the new board Core9558 module, the RF design will according to the reference design and this issue should be fixed.

spiccinini, 2018-12-04:
Fixed in phase2. I will upload the measurments done but for the record sensitivity is similar to tplink Archer AC1750 rev4 (US) in all antenna ports (using same signal generation with USRP and direct cable connections).

closed 2018-12-04 by spiccinini