LibreRouterOrg/board #3: Board hangs after reboot.

nicopace, 2017-11-09:
Reboot without power-cycling doesn't work.

dangowrt, 2017-11-09:
sounds like the SPI chip doesn't get the reset and/or is switched
to stateful 4-byte-address mode (which should be avoided in spi-nor).
Once I get the board to do anything I can check whether JTAG works
after reboot and figure out where the boot hands.

dangowrt, 2017-12-15:
related to #19

tgillett, 2017-12-26:
Addressed by Daniel Golle's work on the SPI chip.
May require a different chip type.
Moved to Phase 2 testing to check design solution.

spiccinini, 2018-01-26:
We have agreed to change the specification of the 2nd prototype to 16MB.

dragino, 2018-03-08:
While using the 32MB flash chip w25q256fv, if we connect the current pin1(HOLD) and pin3(/RESET) together, since pin1 is connected to SYS_RST_L . while the QCA9558 reboot, it will send a low to SYS_RST_L so to reset the flash. This can fix this reset issue.

spiccinini, 2018-11-20:
This is fixed in phase2 (MegaBoard v1.1 / Core v0.2)

closed 2018-12-04 by spiccinini