LibreRouterOrg/board #7: Link LED Operation

nicopace, 2017-11-09:
The Link0 and Link 1 LEDs are only lit briefly at initial power up.

The LINK LEDs are shown on schematic pp 11 as being driven by the following signals:

LINK0 S17_LEDn_3 , S17_LEDn1000_3
LINK 1 S17_LEDn_4 , S17_LEDn1000_4

These signals are referenced on schematic pp 9 but don't appear to be connected to any other device.
They do not appear anywhere else in the schematic.

tgillett, 2017-12-26:
The Link LEDs both light up during the board power up sequence, which indicates that they are correctly connected to the AR8327 chip.
Also I can ground the signal end of the 330R resistors and the LED lights up.

So my conclusion is that the AR8327 is not programmed by default to operate the LEDs, at least not for my test.  The description of the LED control is shown in Sect 2.9 (pp36) of the AR8327 data sheet.
My reading of the default settings is that the LEDs will only light up if the link setting is detected as being a Gigabit link.

When connected to my laptop, the laptop detects the link as being Gigabit, but perhaps the board does not.  This may explain why the LEDs do not light up in my testing, particularly if the wiring mods around the RJ45 connectors are such that the board does not detect the link as being gigabit.

dragino, 2018-01-24:
the first batch prototype has wrong trace for the LEDs ports, also the 100M and 1000M LED can't be simple short together. This will be fixed in the 9558 Mega board v1.0