LibreRouterOrg/board #20: Prototypes have fake and reused MAC addresses

altergui, 2018-05-23:
Compare these two outputs, produced in two different nodes running LibreRouters v1.0 with two HPM5G modules each

```
root@czuk-bbone:~# ethtool -P eth0
Permanent address: 14:3d:f2:cc:92:72
root@czuk-bbone:~# ethtool -P eth1
Permanent address: 14:3d:f2:cc:92:71
root@czuk-bbone:~# ethtool -P wlan0-adhoc
Permanent address: 00:11:7f:13:36:36
root@czuk-bbone:~# ethtool -P wlan1-adhoc
Permanent address: 00:02:03:04:05:06
root@czuk-bbone:~# ethtool -P wlan2-adhoc
Permanent address: 00:02:03:04:05:06
```

```
root@oncelotes-bbone:~# ethtool -P wlan0-ap
Permanent address: 00:11:7f:13:36:36
root@oncelotes-bbone:~# ethtool -P eth0
Permanent address: 14:3d:f2:cc:92:72
root@oncelotes-bbone:~# ethtool -P eth1
Permanent address: 14:3d:f2:cc:92:71
root@oncelotes-bbone:~# ethtool -P wlan0-ap
Permanent address: 00:11:7f:13:36:36
root@oncelotes-bbone:~# ethtool -P wlan1-adhoc
Permanent address: 00:02:03:04:05:06
root@oncelotes-bbone:~# ethtool -P wlan2-adhoc
Permanent address: 00:02:03:04:05:06
```

This makes it impossible to mesh out-of-the-box either via wireless or ethernet, since MAC address on `eth0` of czuk-bbone is identical to the MAC of `eth0` in oncelotes-bbone (and all the other eth0 in other LibreRouters). Same happens for wlan0.

With HPM5G modules the problem gets worse, since it's very unexpected to have a system with two different physical interfaces with identical MAC address. 80211s kernel code warns repeatedly since boot (this was reported upstream https://www.spinics.net/lists/linux-wireless/msg168634.html)

Current workaround we do is to generate random macaddresses on first boot, and hardcode them in uci config, so those are used instead.

darkdrgn2k, 2018-05-24:
I can confirm this also happens when the board is placed in an espresso bin

Each board gets the Mac of
00:02:03:04:05:06

I needed to set a random Mac before trying to use it.

dragino, 2018-05-25:
Solution for puttiing different MAC for ETH0 and ETH1. 
1/ cat /dev/mtd6 > art.bin
2/ transfer the art.bin to PC. it is a bin file, modify the beginning of the file to have different ETH0 and ETH1. 
The offset is:
Eth0: 0x000000
Eth1:  0x000006
wlan0: 0x00000C

modify the data on above address for a different MAC and generate a new art.bin file. 
3/ Overwite this new art.bin to the art.bin in the flash and can solve the ETH0 and ETH1 mac issue. 

In future production, they will be of unique MAC address when made in factory