LibreRouterOrg/board #21: 2.4 Ghz radio issue

nicopace, 2018-05-26:
Hi, 
I open this issue to document the bug of the Qualcomm chipset so we can reach out to find a solution.
Please @dragino could you tell us a more?

sentenwin, 2018-05-26:
@dragino requesting more information to fix the issue

dragino, 2018-05-28:
The RX test is under QSDK from Qualcomm and ART2  tool, hardware is connect the QCA9558 to IQVIEW and test the RX sensitivity. The issue we see are:
1/ The packet lost in channel3 is quite high, it almost 100% lost. 
2/ all three channels' RX in 5L mode test is very poor. 

the log are in the link [rx issue log](http://www.dragino.com/downloads/test/Core9558/rx_issue/)

spiccinini, 2018-05-31:
Hi! I want to participate in the debug of this issue. I have reviewed the schematic design and have some questions. As I don't have the hardware I can't answer them myself.

- [ ]  Is TX working ok? Is the TX output power as expected?
- [ ] Is the LNA transistor (BFR740L3RH) properly powered? Measure with oscilloscope the voltages at the LNA transistor (Q4, Q5, Q6 BFR740L3RH). If the router is in AP mode a 'squarish' wave should be seen. Which are the levels of the wave?
- [ ] Is the  Tx/Rx switch UPG2179TB  property powered and commanded? Measure voltages with oscilloscope. Is the switch of RX/Tx correctly working (in RX mode when not transmitting)?
- [ ] What is the board fabrication specification? Material of the PCB, cooper size and stack spacing. This is needed to check the impedance design of tracks.

Review of all the components of this part of the circuit:
- [ ]  Review the components in the Bill of Materials. Check that every component is the same as the schematic. 
- [ ]  Are all reviewed in the PCB/Gerber level, that pin-1 is pin-1 and pin-N is pin-N and is not mirrored, etc?
- [ ] Are the components in the PCB what they should? Sometimes at fabrication time some components get mixed or incorrect values are placed. 
- [ ] Visual check: Capacitors, resistors, and inductors looking ok?
- [ ] Components value check, aka Measure check. (As may be needed to unsolder some components this should be left as the final item)

Random things/questions
- [ ] TBF2012-2.45GHz component seems to have input and output inverted in the schematic. As it's a passive filter it could be the same but should be double-checked.
- [ ] Why is the TBF2012-2.45GHz filter in parallel with a 0ohm resistor? Is it loaded in the reference design AP135?
- [ ] The application note of the LNA for 2.4GHz designs [AN173](https://www.infineon.com/dgdl/AN173.pdf?fileId=db3a304329a0f6ee0129a6ab05ba2b59) and the AP135 reference design of Qualcomm are different circuits (the circuit seems similar but the values are different, some are no loaded, etc). In the description of changes of the  AP135 schematic are some comments on changes made to the LNA circuit: some optimizations and simplifications, fixing of the stabilization of the circuit, etc. Some of this may be optimizations due to a specific bill of materials (capacitors/inductors with specific properties like R_series, etc) and to a specific PBB material and stack-up. Do we have the Bill of materials, gerber and stack-up of AP135? It could be wise to test the LNA with the values and bill of materials stated in the AN173.

spiccinini, 2018-06-01:
Hi @dragino ! If the rev1 design is the same for this part of the circuit then maybe it's easier if I get a board from @nicoechaniz as I am located in BsAs/Argentina.

Can we add the Bill of Materials and the PCB stackup to this repository? It is very useful information for debugging.
Thanks!

dragino, 2018-06-09:
Update on this issue. We have found the issue. There are two reasons:
1. The calibration software we are using. 
2. The ANT Channel 3 has hardware design issue. 

Now in the calibrate software can set the board as 2T2R and work at 17dB and pass all the tests.

dragino, 2018-06-09:
We plan to release a new version of board which can work with all channels to 20dB as the LibreRouter requirement.

benhylau, 2018-06-09:
Great news @dragino! Is it possible to make this hardware modification for an existing LibreRouter board with basic tools? If not, what are the steps to set the 2.4G radio to 2T2R 17 dB to make that radio somewhat functional?

darkdrgn2k, 2018-07-07:
It seems that meshpoint and adhoc does infact work, however needs to be in EXTREME proximity (cm in distance)

spiccinini, 2018-11-21:
I confirm it's fixed in Core v0.2/MegaBoard v1.1 (phase 2).  
Sensitivity still not measured but over the air tests shows nominal behaviour.

closed 2018-11-21 by spiccinini

spiccinini, 2018-12-04:
measurment notes: https://github.com/libremesh/librerouter/wiki/Wireless-measurements-with-USRP-notes