LibreRouterOrg/board #22: Board hangs when power cycled with serial TTL connected

darkdrgn2k, 2018-07-07:
When attaching power to Prototype V1 board with a serial ttl attached (ground/rx/tx pins) boot seems to hang

```
U-Boot 1.1.4 (Apr 19 2014 - 00:17:22)

ap135 - Scorpion 1.0DRAM:
sri
Scorpion 1.0
ath_ddr_initial_config(233): (32bit) ddr1 init
tap = 0x00000002
Tap (low, high) = (0x4, 0x1e)
Tap values = (0x11, 0x11, 0x11, 0x11)
 4 MB
Top of RAM usable for U-Boot at: 80400000
Reserving 207k for U-Boot at: 803cc000
Reserving 192k for malloc() at: 8039c000
Reserving 44 Bytes for Board Info at: 8039bfd4
Reserving 36 Bytes for Global Data at: 8039bfb0
Reserving 128k for boot params() at: 8037bfb0
Stack Pointer at: 8037bf98
```
Not sure if it is related to https://github.com/libremesh/librerouter/issues/3

(refrenced  https://github.com/tomeshnet/meshstream/issues/7)

spiccinini, 2018-12-04:
With my serial adapter I don't have this problem.

spiccinini, 2018-12-04:
I don't have this problem with V1 nor V2. Probably my ttl adapter has a higher impedance in the RX pin.