LibreRouterOrg/board #23: PoE Passthrough overheats magnetics

spiccinini, 2018-11-21:
T1 magnetics (QT48A03) gets very very hot in one minute when doing PoE passthrough to another LR (total power measured at the power source: 14W ) 

Some notes:
* QT48A03 datasheet doesn't say it is PoE compliant.
* Here are some magnetics that supports PoE http://www.silvertel.com/poe-documents/88-poe-magnetics.html
* PoE (and PoE passthrough) is using only one pair. TODO: Perform the same test with the other pairs in parallel.

dragino, 2018-11-26:
See if R1 /R4/R6/R26 are soldered. They are in the back side of the transformer. If they are there, there will be heat issue. Please see your boards has them there.

spiccinini, 2018-11-27:
Yes, R1, R2, R4, R5, R6, R29, R30 and R31 are all soldered. If my calculations are correct this is seen as a 112ohm load per device. So when PoE is in use the aditional consumption is 1.28W disipated in thouse resistors. And then when PoE passthrough is enabled a second load in parallel of the same value is connected addind an aditional 1.28W  (2.6W total).

I will test unsoldering the resistors, but common mode impedance of the line will suffer without the resistors. I don't how much this affects ethernet.  
I didn't study the PoE circuit  in detail but I think that one capacitor in series with each resistor could be added to cut current flow.

spiccinini, 2018-12-12:
Measurements https://github.com/libremesh/librerouter/wiki/PoEPT-(Pass-through)-heat-vs-voltage

spiccinini, 2018-12-18:
The solution proposed and measured working is:

Disconnect R4 and R5 from C49. Add 2 capacitors 10nF/2kV to the schematic:
MCT2 <–> R4 <–> 10nF/2kV <–> GND
MCT6 <–> R5 <–> 10nF/2kV <–> GND

Make sure there is no DC connection between R4 to R5.

Also R2 doesn't have to be loaded (as is in parallel with R1)

spiccinini, 2019-05-15:
Chage done in MegaBoard v1.2 eagle project

spiccinini, 2019-05-15:
Fixed!

closed 2019-05-15 by spiccinini