LibreRouterOrg/board #25: Gas Discharge overheats if input voltage is >= 30V

spiccinini, 2018-12-03:
There is a problem when voltage go over 30V (with and without PoE). Consumption goes from 4.7W@30V to 7W@32V. The component that get's very hot seems to be V1 BC301N-D Gas Discharge Tube.
So the usable range is 8V to 29V at least in the board I measured.

tgillett, 2018-12-03:
The BC301N-D should have a breakdown voltage of 300V. Below this value it should appear as an open circuit.
Does the component have identifying markings on it?
Perhaps it is an incorrect component in that position.

spiccinini, 2018-12-03:
This is the component that gets really hot (all the area is hot but this is specially hot or it conducts heat a lot more than the others as I am measuring with my finger)
[image]

I will try to unsolder and test again without V1, but yes, you are right that this component shouldn't be the problem.

dragino, 2018-12-04:
Do we need to power it for over 30V?

spiccinini, 2018-12-04:
No, it is not a requirement as far as I know. I opened an issue because the input range was advertised from 8 to 36. 
We can shrink the input range from 8 to 29?, but I think it is interesting to know which component is the culprit (in case we want to extend the range) so we can know for sure the range, I only measured one device and maybe another device has lower input range. 
I didn't have the time to review all the input components datasheets.  

Other important things I noted in  the circuit:
* Input capacitor C76 is 25V rated. This is far from good if we want to allow 24V in the range. Also this is an electrolytic capacitor we should use ceramic or tantallum if posible. Proper derating depends on type of capacitor.

spiccinini, 2018-12-06:
Diode D1 (SMBJ24CA)  is a 24V Transient Voltage Suppression Diode. This should be the culprit but I didn't have the time to test it. We should change this for a SMBJ40CA that is rated 40V

spiccinini, 2018-12-12:
I measured up to 32V of input voltage without heat removing D1. (So BC301N-D Gas Discharge Tube is not the problem)

The proposed changes are:
*  change D1 to SMBJ40CA  (that can handle up to 40V).
* change C76 to a cap of at least 36V

spiccinini, 2019-05-15:
Chages done in MegaBoard v1.2 eagle project.

spiccinini, 2019-05-15:
Fixed!

closed 2019-05-15 by spiccinini