LibreRouterOrg/board #31: GPS module does not lock (signal issues)

spiccinini, 2019-04-15:
The GPS module never locks to the GPS satellite signal when connected to the Mega Board.
If the GPS module works ok if it is powered by an external source.

Main culprit could be power source noise from the RT7272 DC-DC switching modules. This modules oscillate at 500MHz that is nearly 1/3 of the GPS signal frequency. So signal jamming from the power source could be the problem in a mix of conducted or emited noise.

The GPS module has  a ferrite ( PZ2012D800-3R0TF) in series for filtering.

G10h4ck, 2019-04-16:
>The GPS module has a ferrite ( PZ2012D800-3R0TF) in series for filtering.

CodeFetch, 2019-05-03:
Check if the power fluctuates with an oscilloscope. Maybe it just needs a decoupling capacitor.

spiccinini, 2019-05-14:
> Check if the power fluctuates with an oscilloscope. Maybe it just needs a decoupling capacitor.
Thanks @CodeFetch for the sugestion. I already mesured, DC power is good, and the GPS board already has a DC decoupling capacitor and also a ferrite bead in series but it seems it is not enaugh.

spiccinini, 2020-01-13:
The culprit of this issue is that the PCIe is generating the interference. The power sources are not the problem as we bypassed them and the problem persisted. When we disable PCIe directly from the registers of the SoC then the problem dissapears. Also the noise is not present when the board is booting until the PCIe root complex starts.

We are working at #54  improving the impedance of the PCIe lines in the mega board with some hope.

CodeFetch, 2020-01-13:
Some time ago I was told by an electronics engineer that the PCIe lines are far too long and should not move along the whole board. He said even with shielding it would unlikely pass CE conformity tests. I'll kindly ask him to have a look for a solution.