LibreRouterOrg/board #34: If the LR is powered by the PoE pass port some led turns on

spiccinini, 2019-07-01:
We should add a low voltage diode from the PoE to the PoE pass port as some current is going from the PoE_pass that at least powers a LED, being very confusing.

spiccinini, 2019-07-02:
Maybe we can make PoE work in both ports as input but for poe pass only one port is the input.

xcancerberox, 2020-01-06:
Proposal: duplicate the circuit of poe_injection and add diodes to ensure the current circuit.

[image]

xcancerberox, 2020-01-06:
Add hardware switch for default state of the POE injection enable.

[image]

xcancerberox, 2020-01-08:
Added by 0140862596738967a777f8743e4311cc6e76c370 and f2e57b348a968a50a00aaf77966036c01f5a0f3e

closed 2020-01-08 by xcancerberox