LibreRouterOrg/board #51: Complete BOM

xcancerberox, 2019-11-19:
The BOM is not fully defined.

We will copy the components that are defined in the eagle design and define thous that are not.

xcancerberox, 2019-11-26:
Change C6, C57-C59, C75, C110, C111, C116, C124 from "0.1uF" to "0.1uF, 10V".

So we can use same component as C5, C10-C13, C21, C30, C33-C35, C37, C42, C62, C71, C72, C91, C92, C102, C103, C137

xcancerberox, 2019-11-26:
For C40, C41, C49 (10nF, 2KV) I only found 

https://eu.mouser.com/ProductDetail/Yageo/CC1812JKX7RDBB103?qs=tS%2FAHvPQ%2F55AosBJ0y7d0A%3D%3D

spiccinini, 2019-11-26:
We can change the 2kV to a more common value >500V

xcancerberox, 2019-11-26:
Changed to:

https://www.digikey.com/product-detail/en/murata-electronics/GCJ31CR73A103KXJ3L/490-16694-6-ND/7595929

xcancerberox, 2019-11-26:
For  C18, C27, C43, that are defined as Nichicon_6.6x7.7 a taller replacement will be used:

https://www.digikey.com/product-detail/en/nichicon/UCM1A471MCL1GS/493-14524-6-ND/5800270

xcancerberox, 2019-12-01:
Inductors L2-L5, L8 part number lead me to https://www.datasheets360.com/part/detail/swpa6045s6r8mt/-8277091879156552518/

[image]

With thous characteristics I found:

https://www.digikey.com/product-detail/en/abracon-llc/ASPI-6045S-6R8M-T/535-13046-6-ND/5413867

This is ok?

spiccinini, 2019-12-11:
Leftover from  #53:

There are some components that are not defined at all and we need to ask to the designer.
```
    2G_WLAN1, 5G_WLAN1, P4_1000, STATUS1 -> 3mm/GREEN
    P3_100, P4_100, PWR1 -> 3mm/RED
    5G_WLAN0, P3_1000, USB_LED1 -> 3mm/BLUE
    D3, D4, D6-D11 -> BS05C
    EARTH1 -> M4
    P1-P8 -> 3R420L
    S2 -> push_switch_SH
    SIM_CARD_CON_6MICRO0 ->SIM_Card
    T1-> QT48A03
    U12 -> USB_HUB_FE1.1S
    V1-> BC301N-D
```

In the original BOM this components are "domestic / made in china":
` ` 
S2 -> push_switch_SH
SIM_CARD_CON_6MICRO0 ->SIM_Card
EARTH1
All Leds
```

PR #53 (2019-12-11): Mega board bom completion part A
BOM completion using information from EAGLE (really poor).

close #51 

There is some components that are not defined at all and we need to ask to the designer.

* 2G_WLAN1, 5G_WLAN1, P4_1000, STATUS1 -> 3mm/GREEN 
* P3_100, P4_100, PWR1 -> 3mm/RED 
* 5G_WLAN0, P3_1000, USB_LED1 -> 3mm/BLUE 
* D3, D4, D6-D11 -> BS05C 
* EARTH1 -> M4 
* P1-P8 -> 3R420L 
* S2 -> push_switch_SH
* SIM_CARD_CON_6MICRO0 ->SIM_Card 
* T1-> QT48A03 
* U12 -> USB_HUB_FE1.1S
* V1-> BC301N-D