LibreRouterOrg/board #54: Improve impedance of PCIe routing

spiccinini, 2019-12-10:
Differential impedance must be 100ohm. 

Current Mega board calculated impedance is 130ohm.
[image]

One proposed solution:
[image]

xcancerberox, 2019-12-10:
The PCIe connector pad size is 0.6mm. I'm using that for track width and 8mils for separation:

[image]

xcancerberox, 2019-12-17:
Change decoupling capacitor network. Originally two 47uF where close to una pcie and two 33pF capacitor to the other pcie.

[image]

[image]

Separate one of each to each pcie.

xcancerberox, 2019-12-18:
The routing whit the calculated impedance was added on f1ad32cfdfde17e03cad46a9efe0c93327818638

closed 2019-12-18 by spiccinini

xcancerberox, 2020-01-30:
Just for clarification, the final configuration for the differential pairs is:

[image]

PR #56 (2019-12-18): Mega board pcie big move
Change the position of the pcie1 expansion board to improve RF isolation:

* Move PCIe1 to the new position (closes #41 )
* Changes differential pair impedance (closes #54 )
* Fix  power issue #42  (not sure if its ok the change)
* Each PCIe device has a complete (the best that could be done) GND plane at the same side of the device.
* The micro strips for the differential pair were routed trying to avoid cuts on the shield plane