LibreRouterOrg/board #58: Missing holes in PCB

xcancerberox, 2020-01-06:
When we change the design from eagle to kicad some holes disappear.

xcancerberox, 2020-01-08:
Fix on dff2e41a65e24b3435abd1085ba273a77f78e4d1

spiccinini, 2020-01-29:
Just for reference this is how much the gerber files generated by kicad must be translated to match the eagle files
[image]

spiccinini, 2020-01-29:
The four "mounting board holes" are 30 mils below (the blue is the correct position):
[image]

@xcancerberox Maybe something is 30mils in an incorrect position, how did you place them? what was your reference?

spiccinini, 2020-01-29:
The power jack footprint in the kicad design has only circle holes but in the Eagle design has "square" holes in the "edge cuts mask":

[image]

spiccinini, 2020-01-29:
This two problems are the only differences that I can find that have to be fixed.

xcancerberox, 2020-02-03:
On behalf of the jack connector, the holes were defined in the cutout layer that was lost in the kicad translation.