LibreRouterOrg/board #61: Add a way to disconnect the power to the board through GPIO

spiccinini, 2020-02-04:
The idea is to provide a way to cut the power to the LibreRouter.

xcancerberox, 2020-02-05:
Like a self reset? or like suicide?

spiccinini, 2020-02-06:
> Like a self reset? or like suicide?

That functionality is already present through the watchdog that is also available at a GPIO. 

This idea is to provide a way for an external board to use a GPIO ot cut the power of the board for longer periods of time. For example when connected to a solar panel and batteries in the night the LR can be turned on like for 2 minute each 10 minutes saving 80% of the battery power (even detecting if there is a station connected and keeping it on longer periods of time)

xcancerberox, 2020-02-06:
Ok, so the GPIO is an external GPIO.
[image]
We can add there a PMOS transistor like:
[image]
with the same transistors that we are using for POE.